im-tomu/fomu-workshop #17: Serial port doesn't appear on macOS 10.14.6

dan-c-underwood, 2019-08-20:
After successfully loading `micropython-fomu.dfu` onto Fomu (and once the LED starts pulsing red) it does not show up as a `/dev/cu.*` or `/dev/tty.*` device.

No additional serial drivers are installed. Reproduced on two devices - one connecting straight to a USB-A interface, the other to USB-C via an adapter.

xobs, 2019-08-21:
Does it show up as cu.usbmodem? I seem to recall from the previous workshop that the filename changed in recent os versions.

dan-c-underwood, 2019-08-21:
The only cu device that shows up is Bluetooth-Incoming-Port. I have had success with serial devices before (Tomu definitely worked!) so it’s off that this case doesn’t. I’ll check on a different OS to rule out an issue with loading the Fomu firmware.

dan-c-underwood, 2019-08-21:
The device doesn’t show up in Windows either (again - dfu-util works fine). If it helps, the software version of the fomu bootloader is v1.8.7 (received from Crowd Supply).

ret, 2019-08-23:
[Data point, not a solution] Same issue here with the micropython.dfu, but good news is that the device shows up (and the example works) as cu.usbmodem14101 and tty.usbmodem14101, when I load [fomu-workshop/riscv-usb-cdcacm](https://github.com/im-tomu/fomu-workshop/tree/master/riscv-usb-cdcacm). I'm using OSX 0.14.6, and a production fomu. Given the RISC-V example works, does this suggest a problem in the uPy dfu example itself?

ret, 2019-08-24:
Perhaps the new .dfu [micropython-fomu: fix usb_address_write](https://github.com/im-tomu/fomu-workshop/commit/ffadb53856ecff22c52e11d12ba93bdb30afd6f7) fixed this? The device now shows up and micropython works (OSX 10.14.6, old early 2013 mac pro).

```console
$ miniterm.py
--- Available ports:
---  1: ...
---  2: ...
---  3: /dev/cu.usbmodem14201 'Fomu Micropython v1.4.1-6391-g8'
--- Enter port index or full name: 3
--- Miniterm on /dev/cu.usbmodem14201  9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
MicroPython v1.4.1-6392-g573979203-dirty on 2019-08-23; fomu with vexriscv
>>> 1+1
2
>>>```

dan-c-underwood, 2019-08-24:
That looks encouraging! I’ll try it when I’m back home on Monday.

dan-c-underwood, 2019-08-26:
Can confirm that the latest build of `micropython-fomu` has fixed this. Will close this issue now.

closed 2019-08-26 by dan-c-underwood