im-tomu/fomu-workshop #34: Can't open serial port on Linux unless VID/PID are manually bound to serial driver

milkey-mouse, 2019-08-23:
I couldn't get the provided MicroPython firmware to work at first. It seemed to flash fine, but afterwards no `/dev/ttyACM*` were created. I thought I didn't have the right USB serial driver before I realized the USB VID/PID pair for "`Generic Fomu Micropython v1.10-298-g6f94468`" just wasn't registered with any USB serial driver.

Once I figured that out, I manually added it to the `generic` (`ftdi_sio` appears to work too) USB serial driver:

    # lsusb
    Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    ---- 8< ----
    Bus 001 Device 006: ID 1209:5bf0 Generic Fomu Micropython v1.10-298-g6f94468
    # echo 1209 5bf0 > /sys/bus/usb-serial/drivers/generic/new_id

Once the driver picked it up (i.e. it was unplugged and plugged back in) it worked fine.

A hacker's solution would be to use a VID/PID the kernel already knows about (which obviously won't fly on a production release). A better solution would be to get [`1209:5bf0`](http://pid.codes/1209/5BF0/) added to [this list](https://github.com/torvalds/linux/blob/a2d79c7174aeb43b13020dd53d85a7aefdd9f3e5/drivers/usb/serial/ftdi_sio.c#L143-L146). However, other Fomu applications that use the default VID/PID and *don't* emulate a USB serial port might not appreciate this driver taking over by default.

P.S: I've only ever seen `/dev/ttyACM*` when a "real" hardware serial port is behind it; using the USB serial driver with Fomu on Linux results in a device file named `/dev/ttyUSB*`. If that's not just my weird system configuration the instructions should be changed to match.

mithro, 2019-08-23:
`/dev/ttyACM` is provided by the `cdc-acm` driver rather than the `usb-serial` driver. The `cdc-acm` is a [class driver](https://en.wikipedia.org/wiki/Class_driver), so it should be loaded automatically when the USB device is detected.

mithro, 2019-08-23:
@milkey-mouse Can you check if you had the `cdc-acm` driver loaded?

milkey-mouse, 2019-08-23:
Ah, that explains it. I hadn't built that driver for my kernel; I assumed the USB serial drivers were all that would be necessary for a serial port.

closed 2019-08-23 by milkey-mouse