im-tomu/fomu-workshop #35: How to reset to RISCV state after verilog update without unplugging

tatzelbrumm, 2019-08-23:
If I want to jump back from the  
### Hardware Description Languages
section,

is there a way to get the FOMU back to the configuration needed for the
### Python on FOMU
or 
### FOMU as a CPU
chapters without prying the FOMU out of the USB port or turning the host computer on and off?

yorickvP, 2019-08-23:
If you have wishbone, you can reset with `wishbone-tool 0xe0006000 0xac`. If you don't, you'll have to write some verilog to support it. You can also do some usb trickery to reset power to the device.

vshymanskyy, 2019-08-28:
I believe, `wishbone-tool 0xe0006000 0xac` is not working after uploading `riscv-blink`

lromor, 2019-08-29:
Hi! I made a small snippet that reboots the ice40 if you connect the 1-2 pins:

```
module top (
            output rgb0,
            output rgb1,
            output rgb2,
            input  clki,
            input  user_1,
            output user_2,
            output user_3,
            input  user_4,
            );

   assign rgb1 = 1'b1;
   assign rgb2 = 1'b1;

   // Connect to system clock (with buffering)
   wire clkosc;
   SB_GB clk_gb (
      .USER_SIGNAL_TO_GLOBAL_BUFFER(clki),
      .GLOBAL_BUFFER_OUTPUT(clkosc)
   );

   wire clk = clkosc;

   // Configure user pins so that we can detect the user connecting
   // 1-2 or 3-4 with conductive material.
   //
   // We do this by grounding user_2 and user_3, and configuring inputs
   // with pullups on user_1 and user_4.
   localparam SB_IO_TYPE_SIMPLE_INPUT = 6'b000001;

   assign user_2 = 1'b0;
   assign user_3 = 1'b0;

   // Signal receiving the "touch input"
   wire user_1_pulled;
   SB_IO #(
     .PIN_TYPE(SB_IO_TYPE_SIMPLE_INPUT),
     .PULLUP(1'b1)
   ) user_1_io (
      .PACKAGE_PIN(user_1),
      .OUTPUT_ENABLE(1'b0),
      .INPUT_CLK(clk),
      .D_IN_0(user_1_pulled),
   );

   // We connect sc signal to the warmboot.
   // If we touch with conductive material 1-2
   // The fomu reboots.
   SB_WARMBOOT my_warmboot_i (
     .BOOT (~user_1_pulled), //Level-sensitive trigger signal
     .S1 (1'b0), //S1, S0 specify selection of the configuration image
     .S0 (1'b0)
   );

endmodule
```

So a trick could be to use the SB_WARMBOOT (SB=SiliconBlue) (?module?).
You can find more of these modules [here](https://www.latticesemi.com/-/media/LatticeSemi/Documents/TechnicalBriefs/iCETechnologyLibrary.ashx?document_id=44572)

Hope it helps!