im-tomu/fomu-workshop #36: Human readable Makefile versions

tatzelbrumm, 2019-08-23:
I'm not too familiar with make.
I want to read 
`fomu/verilog-blink/Makefile` 
to understand the Verilog-to-uploaded binary design flow.

However, the Makefile contains about 65 lines of variable substitutions, abstractions and generalizations. 
I would appreciate a human readable Makefile with **no variable substitutions** whatsoever, 
so that each line in the Makefile is self-contained, and I do not have to jump to 5 lines of previous definitions, abstractions, and generalizations. 

**Abstraction layers are evil.**
They obfuscate more than they help.

mithro, 2019-08-23:
It is still a work in progress but there should be a more minimal Makefile example at https://github.com/im-tomu/fomu-workshop/blob/master/verilog-blink-minimal/Makefile now. Please take a look and see what you think.

mithro, 2019-08-23:
This makefile is now considered readable. Will log a new issue for more minimal Verilog...

closed 2019-08-23 by mithro