im-tomu/fomu-workshop #43: Make the minimal verilog actually minimal

mithro, 2019-08-23:
Convert the file at https://github.com/im-tomu/fomu-workshop/blob/master/verilog-blink-minimal/blink.v to be more minimal.

 * Remove the usage of the PLL, use a counter instead.
 * Directly drive the pins rather than `SB_RGBA_DRV`

xobs, 2019-08-28:
The blink example now uses a counter rather than the PLL, and does away with the touchpads and pmod stuff.

I feel uncomfortable removing `SB_RGBA_DRV` since directly driving the pins causes an overvoltage on the red LED on production boards, shortening its life.

umarcor, 2020-10-24:
I believe the current examples are minimal enough. As discussed in the chat, the instantiation of `SB_GB` is redundant, because nextpnr will add it anyway. However, I believe keeping it is ok, because it takes 5 lines only, and it's good for didactic purposes. Regarding `SB_RGBA_DRV`, as @xobs said, it wouldn't be good practice to drive the LEDs directly. Therefore, I believe that this issue can be closed.

PR #369 (2020-11-08): docs: rework
Close #39
Close #43
Close #96

This PR includes multiple updates to the docs:

- Several images are converted to figures.
- The white background of multiple images is removed.
- The fomu-block-diagram is redone with draw.io.
- There index, sections Requirements, Background and HDLs are partially rewritten.

See [umarcor.github.io/fomu-workshop](https://umarcor.github.io/fomu-workshop/).
closed 2020-11-08 by xobs