im-tomu/fomu-workshop #50: Covering more topics

vshymanskyy, 2019-08-29:
This is a very nice write-up, thanks! Great explanations, as well as a getting-started guide.
Considering the OSS/OSH aspect of this project, it would be also great to extend this (or possibly organize a separate one) with such ground-up topics:
- How to get the softcore source and how to build it
- How to build `micropython`

P.S. It's a good idea to show how a simple modification/extension can be made in each of the components (like you did). I believe it's an important part for building a healthy developer community around Fomu.

vshymanskyy, 2019-08-29:
I also found some good explanations here: https://forum.mystorm.uk/t/how-fomu-does-things/647
Please consider adding them here.

mithro, 2019-12-27:
@vshymanskyy Any chance you would be able to send us some pull requests? We would gladly merge them. The new sphinx version of the workshop makes it easier to include a lot more detail which people can dive into.