im-tomu/fomu-workshop #54: Fix: evt and prod reference schematics should be valid PDF

tardate, 2019-09-02:
It seems something went wrong with the update of the reference schematics on f5cc2b1188ab0cf3a80cd949b6df26f0510bc6dd. Namely:
https://github.com/im-tomu/fomu-workshop/blob/master/reference/schematic-evt.pdf and https://github.com/im-tomu/fomu-workshop/blob/master/reference/schematic-prod.pdf

The files actually contain an HTML error page "You can’t perform that action at this time." instead of PDF. e.g. https://raw.githubusercontent.com/im-tomu/fomu-workshop/f5cc2b1188ab0cf3a80cd949b6df26f0510bc6dd/reference/schematic-prod.pdf.

Sorry, I'm not yet confident enough to find the right sources to send a PR instead of an issue.

xobs, 2019-09-02:
Thanks!  I can't count the number of times I've accidentally committed a GitHub HTML page rather than the actual file.  But this issue shows that I need to add "2" to that count.

I've replaced the HTML files with their real PDF counterparts.  This should fix the issue!

closed 2019-09-02 by xobs

tardate, 2019-09-02:
Thanks, they look better;-)

mithro, 2019-09-02:
@xobs Is there any reason to include the schematics here rather than linking to their source?

xobs, 2019-09-03:
@mithro The reason they're included here is so that we can hand users a USB drive with everything on it and they can get going.  If we include a link to the source, that requires that they have a good, stable network connection.

tardate, 2019-09-03:
Having the schematics in the workshop repo is certainly useful for getting started without being redirected all over the place. Better yet, perhaps a readme in the reference folder with links to the sources?

commit d091cad4 (2019-09-02): reference: fix pdf files for evt,prod schematic

We had accidentally committed HTML files rather than PDF files.

This fixes #54.