im-tomu/fomu-workshop #55: make error using Fomu Toolchain

dan-c-underwood, 2019-09-06:
Attempting to build either `riscv-blink` or `riscv-usb-cdcacm` using the Fomu Toolchain gives the following error:

`riscv64-unknown-elf-gcc: error trying to exec ‘cc1’: execvp: No such file or directory`

The toolchain bin directory is at the start of $PATH, so it is calling the correct binaries. This is using release v1.3 of said toolchain and is on macOS 10.14.6. Not been able to compare on other platforms yet.

dan-c-underwood, 2019-09-06:
To also add, `make FOMU_REV=pvt1` is also failing with `verilog-blink`. It fails with the error:
`ERROR: proc_share_dirname: unable to determine share/ directory!`

An initial search suggests it’s an issue with the YoSys build - possibly the toolchain build was missing the `YOSYS_DATDIR` parameter.

xobs, 2019-09-06:
The `yosys` error is interesting -- did you move the `bin` directory out from the rest of the toolchain?  Or is there a symlink involved?

I ask because what `yosys` does is attempt to locate itself, then look in `../share/yosys/` for various Verilog library files.  The error you're seeing is indicating that there isn't a `../share/yosys` directory relative to `fomu-toolchain-1.3/bin/yosys`.

The `riscv64-unknown-elf-gcc` error is saying something similar -- it can't locate `cc1`, which ought to exist within the toolchain directory.

dan-c-underwood, 2019-09-06:
I’d managed to do both! I got confused and assumed that the other directories were build artifacts/dependencies.

With `bin` in the same place as the rest of the toolchain, the issue is resolved. Will close! Thanks for the help!

closed 2019-09-06 by dan-c-underwood