im-tomu/fomu-workshop #58: where to get the missing /usr/share/icebox/ice40/chipdb-384.bin

stef, 2019-09-20:
Hi, i'm trying to package all the fomu deps for alpine linux, i managed to compile icestorm, yosys and nextpnr, to verify if everything is ok, i try to run the workshop like this, however when i run `python3 workshop.py --board hacker` i get this:
```
ERROR: Unable to read chipdb /usr/share/icebox/ice40/chipdb-384.bin
0 warnings, 1 error
Traceback (most recent call last):
  File "workshop.py", line 132, in <module>
    main()
  File "workshop.py", line 126, in main
    vns = builder.build()
  File "/home/s/tasks/fomu/fomu-workshop/litex/deps/litex/litex/soc/integration/builder.py", line 180, in build
    toolchain_path=toolchain_path, **kwargs)
  File "/home/s/tasks/fomu/fomu-workshop/litex/deps/litex/litex/soc/integration/soc_core.py", line 461, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/s/tasks/fomu/fomu-workshop/litex/deps/litex/litex/build/lattice/platform.py", line 29, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/s/tasks/fomu/fomu-workshop/litex/deps/litex/litex/build/lattice/icestorm.py", line 183, in build
    _run_script(script)
  File "/home/s/tasks/fomu/fomu-workshop/litex/deps/litex/litex/build/lattice/icestorm.py", line 64, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

i've been looking everywhere but could not find either a matching .bin file, nor a make rule that generates these .bin files. what am i missing?

xobs, 2019-09-20:
Your toolchain is incomplete.  These are generated when `icestorm` is built.  We use scripts derived from @fpgawars to build our static binaries, and those files are copied into the resulting root at https://github.com/xobs/toolchain-icestorm/blob/master/scripts/compile_icestorm.sh#L52

stef, 2019-09-20:
i do have the *.txt files in `/usr/share/icebox/`.

xobs, 2019-09-20:
It sounds like your toolchain was built with external chipdb files, but whoever built it forgot to distribute the chipdb files alongside it.

The mechanism to distribute and build chipdb files isn't documented yet (see https://github.com/YosysHQ/nextpnr/issues/323) so I'm not sure how you'd do that.  I know for `fomu-toolchain` we build the chipdb files into the binary, which is the default.

stef, 2019-09-20:
i've built all of this. icestorm i build like this
```
PREFIX=/usr make
PREFIX=/usr make install
```
after the `make install` the `chipdb-*.txt` files are in `/usr/share/icebox`.

i built nextpnr like this
```cmake -DCMAKE_INSTALL_PREFIX=/usr -DARCH=ice40 -DICEBOX_ROOT=/usr/share/icebox ..
make
make install
```
after building nextpnr i ran `find . -name `chipdb*.bin`, and could not find anything either. however i noticed that the `chipdb*.txt` files have been used as inputs for some transformation.

i didn't use any `-DEXTERNAL_CHIPDB` to nextpnr as you can see.

after a successful build of nextpnr i have these files:
```
ice40/chipdb.py
ice40/chipdbs
ice40/chipdbs/chipdb-1k.bba
ice40/chipdbs/chipdb-1k.cc
ice40/chipdbs/chipdb-384.cc
ice40/chipdbs/chipdb-8k.cc
ice40/chipdbs/chipdb-u4k.cc
ice40/chipdbs/chipdb-5k.bba
ice40/chipdbs/chipdb-5k.cc
ice40/chipdbs/chipdb-8k.bba
ice40/chipdbs/chipdb-u4k.bba
ice40/chipdbs/chipdb-384.bba
ice40/resource/chipdb.rc
build/CMakeFiles/ice40_chipdb.dir/ice40/chipdbs
build/CMakeFiles/ice40_chipdb.dir/ice40/chipdbs/chipdb-1k.cc.o
build/CMakeFiles/ice40_chipdb.dir/ice40/chipdbs/chipdb-384.cc.o
build/CMakeFiles/ice40_chipdb.dir/ice40/chipdbs/chipdb-u4k.cc.o
build/CMakeFiles/ice40_chipdb.dir/ice40/chipdbs/chipdb-8k.cc.o
build/CMakeFiles/ice40_chipdb.dir/ice40/chipdbs/chipdb-5k.cc.o
```

xobs, 2019-09-20:
The `.bba` files get transformed into `.cc` files, which should be included.

However, there are a few things that are really strange:

1. It's looking for `chipdb-384.bin`, when Fomu has a UP5k, so it ought to be looking for `chipdb-5k.bin`
2. The code that prints out that message is only present if you build with `-DEXTERNAL_CHIPDB_ROOT` https://github.com/YosysHQ/nextpnr/blob/master/ice40/arch.cc#L66

Do you have a different version of `nextpnr-ice40` in your PATH that's getting used instead?

stef, 2019-09-20:
you were right. i had an old version of nextpnr on the host. thanks so much for helping figuring that out. it seems i have the fpga toolchain now working for alpine linux! \o/
thanks! <3
s

closed 2019-09-20 by stef