im-tomu/fomu-workshop #59: verilog blink minimal make error

bobwmcgrath, 2019-09-21:
Below is an error I get when trying to make verilog blink minimal.  Regular verilog blink does work however.  What is a compiler directive and why is it missing?  

 Yosys c7f1368cd273f1d84507d29548f3420a08a82702 (Fomu build) (git sha1 ecdc4d8, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)

-- Parsing `blink.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend: blink.v
Parsing Verilog input from `blink.v' to AST representation.
blink.v:119: ERROR: Unimplemented compiler directive or undefined macro `BLUEPWM.
Makefile:14: recipe for target 'blink.json' failed
make: *** [blink.json] Error 1

bobwmcgrath, 2019-09-27:
It works if I copy this over from the other verilog blink

`define BLUEPWM  RGB0PWM
`define GREENPWM RGB1PWM
`define REDPWM   RGB2PWM

xobs, 2019-09-27:
Fixed by #60

closed 2019-09-27 by xobs