im-tomu/fomu-workshop #87: make in riscv-blink doesn't create dfu file

mithro, 2019-12-29:
The `riscv-blink.dfu` is not the default make target like it should be.

ewenmcneill, 2020-02-14:
Looks to me like this has been fixed?

```
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/riscv-blink$ ls
include  ld  Makefile  src
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/riscv-blink$ make
  CC       ./src/time.c	time.o
  CC       ./src/rgb.c	rgb.o
  CC       ./src/usb-dev.c	usb-dev.o
  CC       ./src/usb.c	usb.o
  CC       ./src/main.c	main.o
  CC       ./src/usb-eptri.c	usb-eptri.o
  CC       ./src/usb-epfifo.c	usb-epfifo.o
  AS       ./src/crt0-vexriscv.S	crt0-vexriscv.o
  LD       riscv-blink.elf
  OBJCOPY  riscv-blink.bin
  IHEX     riscv-blink.ihex
  DFU      riscv-blink.dfu
dfu-suffix (dfu-util) 0.9

Copyright 2011-2012 Stefan Schmidt, 2013-2014 Tormod Volden
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Suffix successfully added to file
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/riscv-blink$ ls -l *.dfu
-rwxr-xr-x 1 ewen ewen 4436 Feb 14 21:05 riscv-blink.dfu
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/riscv-blink$ 
```

Ewen

umarcor, 2021-07-10:
I'm closing this since, as @ewenmcneill commented, it seems to be fixed:

https://github.com/im-tomu/fomu-workshop/blob/f045e48144f39ab5064857bd11b0f6ef68222279/.github/tests.sh#L13-L19

closed 2021-07-10 by umarcor