im-tomu/fomu-workshop #92: Handshaking on USB CDC not working correctly

patrislav1, 2019-12-29:
The handshaking on the USB CDC port doesn't work correctly, so when lots of characters are copied / pasted to the terminal at once, then some get lost in the process and don't get echoed.

Reproducible with `picocom /dev/ttyACM0` and the `micropython-fomu.dfu` from the workshop.

tcal-x, 2020-09-20:
Thanks for filing this!   I had noticed this too.   I hit this mostly when I have a small Python program on the host and want to paste it into the terminal window.

I just tried playing around with a few different things.   I found that specifying a slow baud rate for `picocom` _didn't_ help.

I also found that `flterm` works much better.   I was able to paste in entire 5-line programs with no dropped characters.   I happened to have `flterm` in my path from doing a `timvideos/litex-buildenv` build.    I had to add a line-endings option:
```
flterm --line-endings nl2crnl --port /dev/ttyACM0
```
So that's a potential workaround.

xobs, 2020-09-24:
One issue that we should look into is directing people to use Circuit Python instead of Micropython. This is more full-featured and has actual CDC support: https://circuitpython.org/board/fomu/