im-tomu/fomu-workshop #103: Copying code from the tutorial includes "$" prompt

katharosada, 2019-12-30:
On this page: http://workshop.fomu.im/en/latest/python.html

In this box:
[image]

When you click copy, it includes the prompts.

mithro, 2019-12-30:
@katharosada Could you check if this is still happening for you? If so, can you tell me what browser you are using.

katharosada, 2020-01-01:
Still happening on Chrome Version 79.0.3945.88 (Official Build) (64-bit) on MacOSX

ewenmcneill, 2020-02-14:
FTR, still happens for me on Firefox 72 on Linux as well.

See also https://github.com/im-tomu/fomu-workshop/issues/144 which is another user reporting the same issue.  Per https://github.com/im-tomu/fomu-workshop/issues/144#issuecomment-586143280 I think that "click to copy" Javascript is an anti-feature, so perhaps it can just be turned off in the documentation build?

Ewen

eddygta17, 2020-10-06:
@mithro This issue seems to be resolved reading #144 and after testing the copy feature, it works.

closed 2020-10-06 by mithro