im-tomu/fomu-workshop #96: Misunderstanding on HDL page

ktopolski, 2019-12-30:
Hello again,

I was going through the workshop and got confused about the following note:

"The two most common Hardware Description Languages are Verilog and VHDL (the toolchain we are using only supports Verilog)."

Didn't know if I should proceed with "Migen and LiteX" or not, so I asked a question on workshop and got told to open issue on github about misundestanding. So here it goes :).

ktopolski, 2019-12-30:
Forgot to link the page: http://workshop.fomu.im/en/latest/hdl.html

aerohoff, 2020-01-24:
I was able to run the Migen example. Just follow the directions, add the class to the python file after all of the imports, then add the other code after the "soc = BaseSoC..." line. 
Then you can write 3 bit values to the register. Note, my address was different from theirs, so check the csv for the address. 
Each bit corresponds to one of the colors, so 0 is off, 1, 2, 4 will light up one led each, and 7 is all on.

aerohoff, 2020-01-24:
I think by "only supports Verilog", they mean "doesn't support VHDL".

umarcor, 2020-10-24:
The note needs to be updated because the toolchain supports both Verilog and VHDL now. There are mixed HDL examples indeed.

PR #369 (2020-11-08): docs: rework
Close #39
Close #43
Close #96

This PR includes multiple updates to the docs:

- Several images are converted to figures.
- The white background of multiple images is removed.
- The fomu-block-diagram is redone with draw.io.
- There index, sections Requirements, Background and HDLs are partially rewritten.

See [umarcor.github.io/fomu-workshop](https://umarcor.github.io/fomu-workshop/).
closed 2020-11-08 by xobs