im-tomu/fomu-workshop #97: LiteX USB not working

nathanel23, 2019-12-30:
After flashing the litex example via DFU tool the Fomu doesn't show up on USB any more.

00:1a.0 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #4 (rev 03) (prog-if 00 [UHCI])
        Subsystem: Lenovo 82801I (ICH9 Family) USB UHCI Controller
        Flags: bus master, medium devsel, latency 0, IRQ 20
        I/O ports at 1860 [size=32]
        Capabilities: <access denied>
        Kernel driver in use: uhci_hcd
        Kernel modules: uhci_hcd

00:1a.1 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #5 (rev 03) (prog-if 00 [UHCI])
        Subsystem: Lenovo 82801I (ICH9 Family) USB UHCI Controller
        Flags: bus master, medium devsel, latency 0, IRQ 21
        I/O ports at 1880 [size=32]
        Capabilities: <access denied>
        Kernel driver in use: uhci_hcd
        Kernel modules: uhci_hcd

00:1a.2 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #6 (rev 03) (prog-if 00 [UHCI])
        Subsystem: Lenovo 82801I (ICH9 Family) USB UHCI Controller
        Flags: bus master, medium devsel, latency 0, IRQ 22
        I/O ports at 18a0 [size=32]
        Capabilities: <access denied>
        Kernel driver in use: uhci_hcd
        Kernel modules: uhci_hcd

00:1a.7 USB controller: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #2 (rev 03) (prog-if 20 [EHCI])
        Subsystem: Lenovo 82801I (ICH9 Family) USB2 EHCI Controller
        Flags: bus master, medium devsel, latency 0, IRQ 23
        Memory at fc226c00 (32-bit, non-prefetchable) [size=1K]
        Capabilities: <access denied>
        Kernel driver in use: ehci-pci
        Kernel modules: ehci_pci

closed 2020-01-02 by mithro

mithro, 2020-01-02:
@nathanel23 Can you test if this is fixed for you now?

nathanel23, 2020-01-03:
@mithro Yes, just tried it and example is working properly now. Thanks!