im-tomu/fomu-workshop #98: Misunderstanding on Migen and LiteX page

brnbrnrd, 2019-12-30:
This issue concerns the [Migen/LiteX](https://github.com/im-tomu/fomu-workshop/blob/master/docs/migen.rst) page.

It was not clear where I had to copy paste the extra Python code for adding the `fomu_rgb` `csr_register`.

In particular I copy pasted these instructions
```
    # Add the LED driver block
    led_pads = soc.platform.request("rgb_led")
    soc.submodules.fomu_rgb = FomuRGB(led_pads)
```
at the end.

Nothing happened because it is after the building is done.

1. Instructions could be clearer on where to copy-paste the code (also true for the `class FomuRGB`).
2. A warning should tell that I tried to modify registers after they were built.

mainini, 2020-01-03:
IMHO #110 closes this...

xobs, 2020-01-03:
What do you think, @brnbrnrd ?

mithro, 2020-01-03:
I don't think #110 fixes this. It still isn't clear that the code should be added before the builder and there is no error if you try to add something *after* the build has been done.

aerohoff, 2020-01-24:
Yeah, I was able to figure out what to do. But I think it would be helpful to say where to paste the class (after the imports, before main) and where to put the other lines (in main, after the soc = BaseSoC... line), then rebuild and re-dfu. Or maybe there could be comments in the python file saying where they go.