im-tomu/fomu-workshop #121: Import background information about FPGA toolchains from SymbiFlow docs

mithro, 2020-01-08:
I think we can steal a bunch of documentation [from the SymbiFlow docs](https://symbiflow.readthedocs.io/). There is a section called [FPGA Design Flow](https://symbiflow.readthedocs.io/en/latest/toolchain-desc/design-flow.html) that could fit into the [Workshop Background page](https://workshop.fomu.im/en/latest/background.html).

I wonder if there is a way we can just "share" the documentation in some way?

mithro, 2020-01-08:
This is actually one of the reasons I converted to using Sphinx for the workshop.

mithro, 2020-01-08:
@mgielda

umarcor, 2020-10-05:
You can use [intersphinx](https://www.sphinx-doc.org/en/master/usage/extensions/intersphinx.html) for cross-referencing multiple Sphinx sites. For example:

- https://github.com/VUnit/vunit/blob/master/docs/conf.py#L79-L81
- https://github.com/ghdl/ghdl/blob/master/doc/conf.py#L152-L156
- https://github.com/ghdl/ghdl-cosim/blob/master/doc/conf.py#L134-L139