im-tomu/fomu-workshop #123: Sample code for integrating pre-existing verilog module into LiteX build

jnsquire, 2020-01-09:
I've been trying to work out how to integrate existing Verilog modules into the LiteX build, but it's not really clear how everything fits together. Perhaps someone who's worked this out already could create a simple example and add it to the workshop?

mithro, 2020-01-09:
Hi! 

Thank you for the bug report!

There are [some generic documentation in the LiteX-BuildEnv wiki](https://github.com/timvideos/litex-buildenv/wiki).

You might find @bunnie 's "LiteX for Hardware Engineers" page @ https://github.com/timvideos/litex-buildenv/wiki/LiteX-for-Hardware-Engineers useful.

We would definitely welcome a pull request to improve this area!

xobs, 2020-01-10:
The `LED` module is meant to convey how to do this, but it the ability to add an external module.

As @mithro says, we'd welcome some improvement to it!

As a starting point, the previous version of Foboot used the SPI module from `picorv32`.  While the current version of Foboot doesn't use this module, both the verilog source and the LiteX support code are still in foboot, in the form of `hw/rtl/picorvspi.py` and `hw/rtl/spimemio.v`.  The module includes a Wishbone bridge, as well as various LiteX CSRs to expose the module to the host CPU, which would be required in order to make it actually useful.

But the bare minimum can be seen at https://github.com/im-tomu/foboot/blob/master/hw/rtl/picorvspi.py#L96-L126

jnsquire, 2020-01-10:
So the i_ and o_ prefixes get added implicitly... Thanks for the pointers, that helps a lot. I'll see if I can add a simple example myself. Thanks!

closed 2020-01-10 by jnsquire

xobs, 2020-01-10:
There's also io_ for inout, and p_ for parameters.