im-tomu/fomu-workshop #143: tinyusb mass storage example make command doesn't work.

aerohoff, 2020-01-24:
On the bottom of the Fomu as a CPU page, the make command for the tinyusb mass storage example doesn't work. The CROSS_COMPILE arg ends in "-", so maybe there's a typo there?

make command in doc: 
make BOARD=fomu CROSS_COMPILE=riscv64-unknown-elf-

I'm using commits:
workshop 53cffe1761e24e631d79143ff05f36c27bb86885
tinyusb 0d3a7257f54db8b867be35d80b9c1fda85f656d3 

I'm running on macOS if that is relevant.

xobs, 2020-01-27:
Can you give more information?  What error do you see?

The `CROSS_COMPILE` parameter ends in `-` because that's the prefix to the command.  For example, to run `gcc`, the command is prefixed with `riscv64-unknown-elf-`, to give you `riscv64-unknown-elf-gcc`.  If you removed the trailing `-` it would try to run `riscv64-unknown-elfgcc`, which is not the correct name.

jhu960213, 2021-02-23:
Hello @xobs,

I'm following along with the tutorial as well and I've encountered the same problem. I'm running MacOS X 10.15.7 and every time I run this command in the terminal I get this as the output:
 
[image]

These are the steps of the tutorial for the tinyusb:
1) Clone the TinyUSB git repository: git clone https://github.com/hathach/tinyusb (you don’t need to initialize the subrepositories)
2) Change to tinyusb/examples/device/cdc_msc
3) Compile: make BOARD=fomu CROSS_COMPILE=riscv64-unknown-elf-
4) Load it onto the Fomu: dfu-util -D _build/build-fomu/fomu-firmware.bin

So I cloned the tinyusb into my fomu workshop folder and I've also tried cloning it outside of it but it didn't make a difference as I was still getting the same error. What is the cause of the error I'm seeing? Thanks! I'm kinda lost.

J.H

umarcor, 2021-02-24:
@jhu960213 you are calling https://github.com/hathach/tinyusb/blob/master/examples/device/cdc_msc/Makefile, which includes https://github.com/hathach/tinyusb/blob/master/tools/top.mk, where `realpath` is used twice. That is used for defining `TOP`, which is then used in https://github.com/hathach/tinyusb/blob/master/examples/make.mk#L38-L46, where the error is produced.

The point is that `realpath` is not available on macOS by default. You need to either install it or ask maintainers of tinyusb to use a portable alternative.