im-tomu/fomu-workshop #162: docs: Verilog on Fomu docs refer to old directory

ewenmcneill, 2020-02-14:
https://workshop.fomu.im/en/latest/verilog.html (built from https://github.com/im-tomu/fomu-workshop/blob/master/docs/verilog.rst) refers to:

> Enter the `verilog-blink` directory and build the `verilog-blink` demo by using `make`:

but since that was written the https://github.com/im-tomu/fomu-workshop repository has been rearranged and `verilog-blink` at the top level is just a `bin`/`dfu` file (and some intermediate files in a `build` directory) built for some unknown target (the `hacker` and `pvt` boards unfortunately have different LED connectivity, hence the different targets):

```
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/verilog-blink$ ls
blink.bin  blink.dfu  build
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/verilog-blink$ ls build/
blink.asc  blink.json
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/verilog-blink$ 
```

There's no `Makefile` there any more :-)

At minimum the files in that `verilog-blink` directory probably need a README to say what target they were built for (hacker, PVT, ???), and possibly multiple builds for the multiple versions if the idea is that people just install one as a first step?

Presumably this is supposed to now refer to *something* in the `verilog` directory, but I'm unsure what (hence this issue, rather than a pull request).

```
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/verilog$ ls
blink-basic  blink-expanded  pcf
(fomu) ewen@parthenon:/src/fpga/fomu/fomu-workshop/verilog$ 
```

See also https://github.com/im-tomu/fomu-workshop/issues/150 and https://github.com/im-tomu/fomu-workshop/pull/124 which are various other attempts at rearranging those verilog examples.  Whatever is done should include updating the documentation to match where the files ended up :-)   (And after that maybe https://github.com/im-tomu/fomu-workshop/issues/145 can be closed too?)

Ewen

PS: FTR, as mentioned in https://github.com/im-tomu/fomu-workshop/pull/124#discussion_r373922680 and https://github.com/im-tomu/fomu-workshop/pull/124#discussion_r373925643 the `FOMU_REV` via `Makefile` via `yosys` to `ifdef` tests inside the verilog "one file" examples depends on a "newer than `yosys` 0.8" version -- either a build from the git repo in the last 4-5 months (as was packaged for Fomu EVT dev) or `yosys` 0.9 or later.

umarcor, 2020-10-24:
At some point during the last months, these issues were fixed. The links from the docs now point to valid subdirs in the repo, and both `verilog/blink` and `verilog/blink-expanded` contain Makefiles and READMEs. The Yosys version in the toolchain is updated too. Therefore, I think that this issue can be closed.

mithro, 2020-10-24:
Awesome work @umarcor!

closed 2020-10-24 by mithro