im-tomu/fomu-workshop #166: Can't "make env" on MacOSX

mgastel, 2020-02-20:
Using:   MacOSX Catalina 10.15.3

When running make env you are downloading Miniconda3-latest-Linux-x86_64.sh.  On a Mac I believe you should be downloading Miniconda3-latest-MacOSX-x86_64.sh

After changing the Makefile to download the different version of Miniconda the process terminated with the following messages:
```bash
~/d/fomu-workshop   *  docs  make env                                                                                    
mkdir _download
wget https://repo.anaconda.com/miniconda/Miniconda3-latest-MacOSX-x86_64.sh -O _download/Miniconda3-latest-MacOSX-x86_64.sh
--2020-02-19 17:24:39--  https://repo.anaconda.com/miniconda/Miniconda3-latest-MacOSX-x86_64.sh
Resolving repo.anaconda.com (repo.anaconda.com)... 104.16.131.3, 104.16.130.3
Connecting to repo.anaconda.com (repo.anaconda.com)|104.16.131.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 51753385 (49M) [application/x-sh]
Saving to: ‘_download/Miniconda3-latest-MacOSX-x86_64.sh’

_download/Miniconda3-latest-MacOSX-x86_ 100%[============================================================================>]  49.36M  33.8MB/s    in 1.5s

2020-02-19 17:24:41 (33.8 MB/s) - ‘_download/Miniconda3-latest-MacOSX-x86_64.sh’ saved [51753385/51753385]

chmod a+x _download/Miniconda3-latest-MacOSX-x86_64.sh
rm -rf env
./_download/Miniconda3-latest-MacOSX-x86_64.sh -p /Users/mgastel/dev/fomu-workshop/docs/env -b -f
PREFIX=/Users/mgastel/dev/fomu-workshop/docs/env
Unpacking payload ...
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /Users/mgastel/dev/fomu-workshop/docs/env

  added / updated specs:
    - asn1crypto==1.2.0=py37_0
    - ca-certificates==2019.10.16=0
    - certifi==2019.9.11=py37_0
    - cffi==1.13.0=py37hb5b8e2f_0
    - chardet==3.0.4=py37_1003
    - conda-package-handling==1.6.0=py37h1de35cc_0
    - conda==4.7.12=py37_0
    - cryptography==2.8=py37ha12b0ac_0
    - idna==2.8=py37_0
    - libcxx==4.0.1=hcfea43d_1
    - libcxxabi==4.0.1=hcfea43d_1
    - libedit==3.1.20181209=hb402a30_0
    - libffi==3.2.1=h475c297_4
    - ncurses==6.1=h0a44026_1
    - openssl==1.1.1d=h1de35cc_3
    - pip==19.3.1=py37_0
    - pycosat==0.6.3=py37h1de35cc_0
    - pycparser==2.19=py37_0
    - pyopenssl==19.0.0=py37_0
    - pysocks==1.7.1=py37_0
    - python.app==2=py37_9
    - python==3.7.4=h359304d_1
    - readline==7.0=h1de35cc_5
    - requests==2.22.0=py37_0
    - ruamel_yaml==0.15.46=py37h1de35cc_0
    - setuptools==41.4.0=py37_0
    - six==1.12.0=py37_0
    - sqlite==3.30.0=ha441bb4_0
    - tk==8.6.8=ha441bb4_0
    - tqdm==4.36.1=py_0
    - urllib3==1.24.2=py37_0
    - wheel==0.33.6=py37_0
    - xz==5.2.4=h1de35cc_4
    - yaml==0.1.7=hc338f04_2
    - zlib==1.2.11=h1de35cc_3

The following NEW packages will be INSTALLED:

  asn1crypto         pkgs/main/osx-64::asn1crypto-1.2.0-py37_0
  ca-certificates    pkgs/main/osx-64::ca-certificates-2019.10.16-0
  certifi            pkgs/main/osx-64::certifi-2019.9.11-py37_0
  cffi               pkgs/main/osx-64::cffi-1.13.0-py37hb5b8e2f_0
  chardet            pkgs/main/osx-64::chardet-3.0.4-py37_1003
  conda              pkgs/main/osx-64::conda-4.7.12-py37_0
  conda-package-han~ pkgs/main/osx-64::conda-package-handling-1.6.0-py37h1de35cc_0
  cryptography       pkgs/main/osx-64::cryptography-2.8-py37ha12b0ac_0
  idna               pkgs/main/osx-64::idna-2.8-py37_0
  libcxx             pkgs/main/osx-64::libcxx-4.0.1-hcfea43d_1
  libcxxabi          pkgs/main/osx-64::libcxxabi-4.0.1-hcfea43d_1
  libedit            pkgs/main/osx-64::libedit-3.1.20181209-hb402a30_0
  libffi             pkgs/main/osx-64::libffi-3.2.1-h475c297_4
  ncurses            pkgs/main/osx-64::ncurses-6.1-h0a44026_1
  openssl            pkgs/main/osx-64::openssl-1.1.1d-h1de35cc_3
  pip                pkgs/main/osx-64::pip-19.3.1-py37_0
  pycosat            pkgs/main/osx-64::pycosat-0.6.3-py37h1de35cc_0
  pycparser          pkgs/main/osx-64::pycparser-2.19-py37_0
  pyopenssl          pkgs/main/osx-64::pyopenssl-19.0.0-py37_0
  pysocks            pkgs/main/osx-64::pysocks-1.7.1-py37_0
  python             pkgs/main/osx-64::python-3.7.4-h359304d_1
  python.app         pkgs/main/osx-64::python.app-2-py37_9
  readline           pkgs/main/osx-64::readline-7.0-h1de35cc_5
  requests           pkgs/main/osx-64::requests-2.22.0-py37_0
  ruamel_yaml        pkgs/main/osx-64::ruamel_yaml-0.15.46-py37h1de35cc_0
  setuptools         pkgs/main/osx-64::setuptools-41.4.0-py37_0
  six                pkgs/main/osx-64::six-1.12.0-py37_0
  sqlite             pkgs/main/osx-64::sqlite-3.30.0-ha441bb4_0
  tk                 pkgs/main/osx-64::tk-8.6.8-ha441bb4_0
  tqdm               pkgs/main/noarch::tqdm-4.36.1-py_0
  urllib3            pkgs/main/osx-64::urllib3-1.24.2-py37_0
  wheel              pkgs/main/osx-64::wheel-0.33.6-py37_0
  xz                 pkgs/main/osx-64::xz-5.2.4-h1de35cc_4
  yaml               pkgs/main/osx-64::yaml-0.1.7-hc338f04_2
  zlib               pkgs/main/osx-64::zlib-1.2.11-h1de35cc_3

Preparing transaction: done
Executing transaction: done
installation finished.
source env/bin/activate; conda config --system --add envs_dirs /Users/mgastel/dev/fomu-workshop/docs/env/envs
source env/bin/activate; conda config --system --add pkgs_dirs /Users/mgastel/dev/fomu-workshop/docs/_download/pkgs
source env/bin/activate; conda env update --name base --file ./environment.yml
Collecting package metadata (repodata.json): done
Solving environment: failed

ResolvePackageNotFound:
  - yosys
  - netlistsvg

make: *** [env] Error 1
```

mithro, 2020-02-20:
Generation of the docs not on Linux is currently unsupported. You can download a copy of them from workshop.fomu.im

We are working on fixing this, but it will be a little bit.

mgastel, 2020-02-20:
okay thanks.... perhaps a quick update to the main readme would be in order since it says "On MacOS and Linux".
Cheers