im-tomu/fomu-workshop #180: litex/tools/litex_server.py: address already in use

pdp7, 2020-03-02:
From section https://workshop.fomu.im/en/latest/renode-bridge.html
> After this preparation, we are ready to start the server:
> `python3 litex/tools/litex_server.py --usb --usb-vid 0x1209 --usb-pid 0x5bf0`
I get error:
```
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ python3 litex/tools/litex_server.py --usb --usb-vid 0x1209 --usb-pid 0x5bf0
LiteX remote server
[CommUSB] vid: 0x1209 / pid: 0x5bf0 / Traceback (most recent call last):
  File "litex/tools/litex_server.py", line 198, in <module>
    main()
  File "litex/tools/litex_server.py", line 189, in main
    server.open()
  File "litex/tools/litex_server.py", line 36, in open
    self.socket.bind((self.bind_ip, self.bind_port))
OSError: [Errno 98] Address already in use
```
This is me running it again so I think the git errors are just because it already exists:
```
pdp7@x1:~/dev/fpga/fomu-workshop$ cd litex/deps/litex
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ git checkout master && git pull # this will fetch newer LiteX, required to handle communication properly
M	litex/soc/cores/cpu/minerva/verilog
M	litex/soc/cores/cpu/vexriscv/verilog
M	litex/tools/litex_server.py
Already on 'master'
Your branch is up to date with 'origin/master'.
remote: Enumerating objects: 106, done.
remote: Counting objects: 100% (92/92), done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 62 (delta 47), reused 54 (delta 40), pack-reused 0
Unpacking objects: 100% (62/62), done.
From https://github.com/enjoy-digital/litex
   f27225c2..98e41e2e  master     -> origin/master
Updating f27225c2..98e41e2e
Fast-forward
 litex/boards/targets/nexys4ddr.py       | 17 ++++++++++++-----
 litex/soc/cores/cpu/blackparrot/core.py |  2 +-
 litex/soc/cores/cpu/minerva/core.py     |  2 +-
 litex/soc/cores/cpu/picorv32/core.py    |  2 +-
 litex/soc/cores/cpu/rocket/core.py      |  2 +-
 litex/soc/cores/cpu/vexriscv/core.py    |  2 +-
 litex/soc/integration/soc.py            |  6 ++++--
 litex/soc/integration/soc_core.py       | 35 +++++++++++++++++++----------------
 litex/soc/integration/soc_sdram.py      |  7 ++++---
 9 files changed, 44 insertions(+), 31 deletions(-)
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ ./litex_setup.py init  # this will clone dependencies
[cloning migen]...
fatal: destination path 'migen' already exists and is not an empty directory.
[cloning litex]...
fatal: destination path 'litex' already exists and is not an empty directory.
[cloning liteeth]...
fatal: destination path 'liteeth' already exists and is not an empty directory.
[cloning litedram]...
fatal: destination path 'litedram' already exists and is not an empty directory.
[cloning litepcie]...
fatal: destination path 'litepcie' already exists and is not an empty directory.
[cloning litesata]...
fatal: destination path 'litesata' already exists and is not an empty directory.
[cloning litesdcard]...
fatal: destination path 'litesdcard' already exists and is not an empty directory.
[cloning liteiclink]...
fatal: destination path 'liteiclink' already exists and is not an empty directory.
[cloning litevideo]...
fatal: destination path 'litevideo' already exists and is not an empty directory.
[cloning litescope]...
fatal: destination path 'litescope' already exists and is not an empty directory.
[cloning litejesd204b]...
fatal: destination path 'litejesd204b' already exists and is not an empty directory.
[cloning litex-boards]...
fatal: destination path 'litex-boards' already exists and is not an empty directory.
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ export PYTHONPATH=`pwd`:`pwd`/litex:`pwd`/migen
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ 
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ python3 litex/tools/litex_server.py --usb --usb-vid 0x1209 --usb-pid 0x5bf0
LiteX remote server
[CommUSB] vid: 0x1209 / pid: 0x5bf0 / Traceback (most recent call last):
  File "litex/tools/litex_server.py", line 198, in <module>
    main()
  File "litex/tools/litex_server.py", line 189, in main
    server.open()
  File "litex/tools/litex_server.py", line 36, in open
    self.socket.bind((self.bind_ip, self.bind_port))
OSError: [Errno 98] Address already in use
```
I think the local modifications are from the script:
```diff
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ git diff
diff --git a/litex/soc/cores/cpu/minerva/verilog b/litex/soc/cores/cpu/minerva/verilog
index fb296e4e..05186211 160000
--- a/litex/soc/cores/cpu/minerva/verilog
+++ b/litex/soc/cores/cpu/minerva/verilog
@@ -1 +1 @@
-Subproject commit fb296e4e48e5ced8dd05f2228d84b4bc18f54f75
+Subproject commit 0518621134dd7a551cac5758d9171a24d0c1578f
diff --git a/litex/soc/cores/cpu/vexriscv/verilog b/litex/soc/cores/cpu/vexriscv/verilog
index 8baad219..854f9bd2 160000
--- a/litex/soc/cores/cpu/vexriscv/verilog
+++ b/litex/soc/cores/cpu/vexriscv/verilog
@@ -1 +1 @@
-Subproject commit 8baad219885a47f65959a9cd4870691e84678db4
+Subproject commit 854f9bd2282c97251ce65e4117c5cf1630722004
diff --git a/litex/tools/litex_server.py b/litex/tools/litex_server.py
index e17fe115..78a79ff8 100755
--- a/litex/tools/litex_server.py
+++ b/litex/tools/litex_server.py
@@ -17,7 +17,7 @@ from litex.tools.remote.etherbone import EtherboneIPC
 
 
 class RemoteServer(EtherboneIPC):
-    def __init__(self, comm, bind_ip, bind_port=1234):
+    def __init__(self, comm, bind_ip, bind_port=12345):
         self.comm = comm
         self.bind_ip = bind_ip
         self.bind_port = bind_port
```

pdp7, 2020-03-02:
Note: I am using Ubuntu 19.10

pdp7, 2020-03-02:
Turns out it was already running somehow:
```
root@x1:/home/pdp7/dev/fpga/fomu-workshop/litex/deps/litex# netstat -nalp |grep 1234
tcp        0      0 127.0.0.1:1234          0.0.0.0:*               LISTEN      30076/wishbone-tool
```
I killed it:
```
root@x1:/home/pdp7/dev/fpga/fomu-workshop/litex/deps/litex# kill 30076
root@x1:/home/pdp7/dev/fpga/fomu-workshop/litex/deps/litex# netstat -nalp |grep 1234
tcp6       0      0 2001:16b8:5c63:51:41234 2a00:1450:4001:818::443 ESTABLISHED 19458/chrome --type 
root@x1:/home/pdp7/dev/fpga/fomu-workshop/litex/deps/litex# logout
pdp7@x1:~/dev/fpga/fomu-workshop/litex/deps/litex$ python3 litex/tools/litex_server.py --usb --usb-vid 0x1209 --usb-pid 0x5bf0
LiteX remote server
[CommUSB] vid: 0x1209 / pid: 0x5bf0 / tcp port: 1234
```

xobs, 2020-03-02:
This is something that probably should be addressed.

`litex_server.py` is a pure-Python version of the server, and is the original.  `wishbone-tool` is a reimplementation of that functionality, plus a whole lot more.  We could remove the entire "Setting up the server" section and just tell people to run:

```sh
$ wishbone-tool -s wishbone
```

xobs, 2020-03-02:
Does 2707926e8d83e144b7ee9c6346f7eb9d9a54d198 address the issue?

pdp7, 2020-03-02:
It does seem simpler to run:
```wishbone-tool -s wishbone```
The example of running 'led_toggle' in Zephyr shell still works ok with this commnad:
```
uart:~$ led_toggle
uart:~$ led_breathe
```
So I think this is an improvement

pdp7, 2020-03-02:
@xobs I think this is OK now, thanks

closed 2020-03-02 by pdp7