im-tomu/fomu-workshop #182: enabled logging for sysbus.led

pdp7, 2020-03-02:
In section: [Wishbone bridge between Renode and Fomu
](https://workshop.fomu.im/en/latest/renode-bridge.html
)
>The script you loaded configures Renode to log all communication with Fomu. After issuing some commands in Zephyr’s shell you’ll see:
> ```
> 01:00:31.8276 [DEBUG] led: [cpu: 0x40000988] WriteUInt32 to 0x8 (unknown), value 0x7.
> 01:00:31.8279 [DEBUG] led: [cpu: 0x40000990] WriteUInt32 to 0x4 (unknown), value 0x8.
> ```
However, I see no log output unless I run `sysbus LogPeripheralAccess sysbus.led`:
```
(monitor) include @scripts/complex/fomu/renode_etherbone_fomu.resc
(machine-0) start
Starting emulation...
(machine-0) sysbus LogPeripheralAccess sysbus.led
```
[image]

xobs, 2020-03-02:
Thanks for the patch.  In fda6bed5ca802c5b7fd9a8e267316bbbbaf9badf we now tell users to enable logging, as per your instructions.

pdp7, 2020-03-02:
thanks @xobs

closed 2020-03-02 by pdp7