im-tomu/fomu-workshop #266: Broken links related to _static

tcal-x, 2020-07-18:
On the page
https://workshop.fomu.im/en/latest/python.html
(https://github.com/im-tomu/fomu-workshop/blob/master/docs/python.rst)

Link "...(to be found for instance in [riscv-blink](https://workshop.fomu.im/en/latest/_static/riscv-blink/include/generated/csr.h))" points to "https://workshop.fomu.im/en/latest/_static/riscv-blink/include/generated/csr.h".

Link "...can be found in the [ICE40 LED Driver Usage Guide](https://github.com/im-tomu/fomu-workshop/blob/master/docs/_static/reference/FPGA-TN-1288-ICE40LEDDriverUsageGuide.pdf)." points to "https://github.com/im-tomu/fomu-workshop/blob/master/docs/_static/reference/FPGA-TN-1288-ICE40LEDDriverUsageGuide.pdf".

tcal-x, 2020-10-15:
The link for ICE40 LED Driver now works, but the "riscv-blink" link is still broken.

umarcor, 2020-11-08:
This was fixed in #350.

closed 2020-11-08 by umarcor

tcal-x, 2020-11-08:
> This was fixed in #350.

Thanks @umarcor !