im-tomu/fomu-workshop #318: fomu looses programming

arblake, 2020-10-05:
I have uploaded the code using dfu -D build.dfu. Success and reported that ttyACM0 is available. dmesg also reports ttyACM0 is available. Attempts to communicate with minicom fail. device /dev/ttyACM0 no longer present. see below

Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%       104090 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
.......
new full-speed USB device number 10 using xhci_hcd
[ 6706.026844] usb 1-3.4: New USB device found, idVendor=1d50, idProduct=6130, bcdDevice= 0.00
[ 6706.026853] usb 1-3.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 6706.688830] cdc_acm 1-3.4:1.0: ttyACM0: USB ACM device
........
minicom: cannot open /dev/ttyACM0: Input/output error
........
new full-speed USB device number 14 using xhci_hcd
[ 7687.769675] usb 1-3.4: Device not responding to setup address.
[ 7687.977344] usb 1-3.4: Device not responding to setup address.
[ 7688.185311] usb 1-3.4: device not accepting address 14, error -71
[ 7688.186262] usb 1-3-port4: unable to enumerate USB device
...
any suggestions welcome.

arblake, 2020-10-13:
> Hi @arblake <https://github.com/arblake> , is this with the
> micropython example?

No,  FOMU  pvt  trying to load eforth

Adrian

-- 
Adrian Blake, VK2ALF
101 Mulach St
Cooma, NSW, 2630
Australia
Mobile +61 407232978

Pärna tn 3
Otepää
Valgamaa 67404
Estonia
Mobile +372 51971441

rob-ng15, 2020-10-13:
> Maybe @rob-ng15 <https://github.com/rob-ng15> has advice for debugging
> the hw/sw USB stack used by eforth?
>
> Another thing you can try looking at: before programming, try running dfu-util
> -l a few times. Hopefully you'll always see
>
> Found DFU: [1209:5bf0] ver=0101, devnum=12, cfg=1, intf=0, path="1-2.1", alt=0, name="Fomu PVT running DFU Bootloader v2.0.3", serial="UNKNOWN"
>
> If you sometimes see Failed to retrieve string descriptor 2 and
> name="UNKNOWN", you may be seeing an issue I've observed but haven't
> diagnosed.

Sorry, it was a borrowed USB stack, and not experienced enough in verilog
to assist in debugging.

tcal-x, 2020-10-14:
@arblake, can you connect after loading the micropython executable?   It also has a TTY USB interface, although it has some limitations (see issue #92 ).

arblake, 2020-10-14:
loaded micropython:
Copying data from PC to DFU device
Download	[=========================] 100%       136164 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Resetting USB to switch back to runtime mode
Done!
Unable to connect to tty interface ... no ttyACM0 or similar

tcal-x, 2020-10-14:
Interesting, I don't get the line `Resetting USB to switch back to runtime mode`.

tcal-x, 2020-10-14:
Ok, a few more questions:

* Which OS are you using?
* Which foboot version are you using (use `dfu-util -l` to see)?
* Did you have to do the [udev stuff ](https://workshop.fomu.im/en/latest/requirements.html#setup-udev-rules)?   This might or might not be necessary.

arblake, 2020-10-14:
Ubuntu , latest
foboot v2.0.2
udev stuff done

arblake, 2020-10-14:
upgraded to foboot v2.0.3
I cannot remove the device and restart micropython. I needs to be reloaded

arblake, 2020-10-14:
If I use Rob's  j1eforth, gtkterm needs to be started before loading software and I cannot stop gtkterm then restart and expect it to communicate with j1eforth

xobs, 2020-10-15:
I'm guessing the problem with `j1eforth` is due to it not meeting timing: https://github.com/rob-ng15/Silice-Playground/issues/8 -- when a design doesn't meet timing, it tends to fail in mysterious ways such as you're seeing. If you try a different Fomu it could work, but since it's running it way outside of spec, it's not guaranteed to work.

For the micropython issue, this sounds like it's as-designed. With Fomu, it runs the bootloader every time you plug it in. To run a program, you either need to load it using `dfu-util -D [file.dfu]`, or start the program that's already loaded with `dfu-util -e`

rob-ng15, 2021-08-30:
I have redesigned and rebuilt j1eforth for the FOMU and have it up and running on my HACKER board. I have built the bitstream for PVT, but I am unable to test.

umarcor, 2021-08-30:
@rob-ng15 maybe you can update or recreate #310?

rob-ng15, 2021-08-30:
I'm sticking with Silice version at the moment, as that I understand!

umarcor, 2021-08-30:
My bad! :wink: