im-tomu/fomu-workshop #319: What's the difference between a regular Yosys and a 'Fomu build'?

umarcor, 2020-10-05:
In section [Required Software](https://workshop.fomu.im/en/latest/requirements.html#required-software):

> Ensure it says **(Fomu build)**. Type `exit` to quit `yosys`.

I'd like to try using Docker image `ghdl/synth:beta`, which includes Yosys and GHDL. However, I am unsure about what needs to be modified in Yosys for supporting Fomu. The procedure in https://github.com/xobs/toolchain-icestorm/blob/master/scripts/compile_yosys.sh seems normal. @xobs are requirements listed somewhere?

mithro, 2020-10-05:
@umarcor -- There is no difference except making sure that you have not installed a random Yosys from a random location. It was important when running workshops with a 1 instruction to 50 student ratio.

umarcor, 2020-10-05:
@mithro, thanks a lot for clarifying. I understand the motivation and I think it makes sense. I was just worried about breaking my loved Fomu if I did something wrong.

I was told about https://github.com/open-tool-forge/fpga-toolchain in the IRC, just a few minutes ago. I will try both approaches, as I'm trying to add VHDL examples to this repo.

closed 2020-10-05 by umarcor