im-tomu/fomu-workshop #331: code works on hacker device not pvt

arblake, 2020-10-07:
I have code that works on the hacker device but not the pvt device. Any  clues why?

mithro, 2020-10-07:
The hacker and pvt devices have different pin out configurations and a few other small changes.

mithro, 2020-10-07:
See the schematic link from https://workshop.fomu.im/en/latest/requirements.html#required-hardware

arblake, 2020-10-07:
This is the error now:
Resetting USB to switch back to runtime mode
error resetting after download
Done!

tcal-x, 2020-10-15:
Hi @arblake , what is the exact sequence you are performing?

arblake, 2020-10-16:
Here:
Start gtkterm try to set to port /dev/ttyACM0 ... it will fail
Insert fpga
dfu-util -e
working
You cannot close gtkterm and restart it   "no port found /dev/ttyACM0". you must remove and reinsert the fpga and issue dfu-util -e