im-tomu/fomu-workshop #346: MicroPython rgb.read_raw() doesn't seem to work properly

sapperlott, 2020-10-15:
When I try to read the LEDDPWR registers from MicroPython, I don't get the expected values:
```
>>> import fomu
>>> rgb = fomu.rgb()
>>> ADDR_RED_LED_PULSE_WIDTH   = 0b0001
>>> ADDR_GREEN_LED_PULSE_WIDTH = 0b0010
>>> ADDR_BLUE_LED_PULSE_WIDTH  = 0b0011
>>> rgb.write_raw(ADDR_RED_LED_PULSE_WIDTH, 10)
>>> rgb.write_raw(ADDR_GREEN_LED_PULSE_WIDTH, 20)
>>> rgb.write_raw(ADDR_BLUE_LED_PULSE_WIDTH, 30)
>>> rgb.read_raw(ADDR_RED_LED_PULSE_WIDTH)
30
>>> rgb.read_raw(ADDR_GREEN_LED_PULSE_WIDTH)
30
>>> rgb.read_raw(ADDR_BLUE_LED_PULSE_WIDTH)
30
```

On closer inspection this doesn't necessarily seem to be a MicroPython issue since I also can't get this to work when using wishbone-tool (trying to read back the values written via MicroPython before):
```
$ wishbone-tool 0xe0006804 1
INFO [wishbone_tool::bridge::usb] opened USB device device 007 on bus 001
Exited MemoryAccess thread
$ wishbone-tool 0xe0006800
INFO [wishbone_tool::bridge::usb] opened USB device device 007 on bus 001
Value at e0006800: 0000001e
Exited MemoryAccess thread
$ wishbone-tool 0xe0006804 2
INFO [wishbone_tool::bridge::usb] opened USB device device 007 on bus 001
Exited MemoryAccess thread
$ wishbone-tool 0xe0006800
INFO [wishbone_tool::bridge::usb] opened USB device device 007 on bus 001
Value at e0006800: 0000001e
Exited MemoryAccess thread
$ wishbone-tool 0xe0006804 3
INFO [wishbone_tool::bridge::usb] opened USB device device 007 on bus 001
Exited MemoryAccess thread
$ wishbone-tool 0xe0006804
INFO [wishbone_tool::bridge::usb] opened USB device device 007 on bus 001
Value at e0006804: 00000003
Exited MemoryAccess thread
```

xobs, 2020-10-15:
Those registers are read-only, not read-write.

The documentation for the RGB block is at https://rm.fomu.im/rgb.html which has a link to the Lattice documentation for the hardware block you're controlling. Appendix D (starting on page 38) documents the actual register set, which has this summary:

[image]

Note the "Access" column that indicates write-only access -- there's no way to read values back.

umarcor, 2020-10-15:
SInce almost all the examples rely on the LED driver, I think it would be interesting to add @xobs' clarification to the docs.

sapperlott, 2020-10-15:
Ah thanks - that clarifies things. But since all the RGB LED registers seem to be write only, wouldn't it make sense to then remove rgb.read_raw() from the Fomu MicroPython library? That would help prevent future users getting confused about this.

xobs, 2020-10-16:
It's only the hard IP that's write-only. Other registers, namely those that wrap the IP and provide control, are read-write. For example, https://rm.fomu.im/rgb.html#rgb-ctrl is readable.