im-tomu/fomu-workshop #412: vhdl/blink/blink.json missing

marcmerlin, 2021-01-03:
/home/merlin/fpga/fomu-workshop/vhdl/blink
```
sauron [mc]$ make
yosys -D PVT=1 \
	-p \
	"ghdl --std=08 ../sb_ice40_components.vhd blink.vhd -e; \
	synth_ice40 \
	-top Fomu_Blink \
	-json blink.json" 2>&1 | tee yosys-report.txt
ERROR: No such command: ghdl (type 'help' for a command overview)
nextpnr-ice40 \
	--up5k --package uwg30 \
	--pcf ../../pcf/fomu-pvt.pcf \
	--json blink.json \
	--asc blink.asc
ERROR: Failed to open JSON file 'blink.json'.
```

That file is missing:
https://github.com/im-tomu/fomu-workshop/tree/master/vhdl/blink

Easy to fix:
```
sauron [mc]$ ln -snf ../../verilog/blink/blink.json blink.json
```

Is there a better fix you'd like to apply?

umarcor, 2021-01-03:
Hi @marcmerlin! It's important to read errors in order. After the first one, all others might be misleading. In this case, the first error is the following:

```
ERROR: No such command: ghdl (type 'help' for a command overview)
```

That means the Yosys you are using was not built with ghdl-yosys-plugin as a VHDL frontend. As a result, Yosys fails, and the `blink.json` is not generated. Then, the next step (second error) is complaining about that.

Can you please let us know how did you install Yosys and which is your OS?

The fix you suggest is not a valid solution. The file `verilog/blink/blink.json` was created by Yosys when you tried example `verilog/blink`. Therefore, it corresponds to the design described in Verilog sources. By symlinking that to the VHDL subdir, you are effectively ignoring VHDL sources. You are running place and route on exactly the same netlist from the verilog example. You don't want to do that. You want to run `verilog`, `vhdl` or `mixed-hdl` examples as isolated projects.

marcmerlin, 2021-01-03:
My bad, you are entirely correct, well crap:
```
sauron [mc]$ yosys
(...)
 Yosys 0.9 (git sha1 1979e0b)
sauron [mc]$ type yosys
yosys is hashed (/usr/bin/yosys)
sauron [mc]$ sudo dpkg --purge yosys
[sudo] password for merlin: 
(Reading database ... 283976 files and directories currently installed.)
Removing yosys (0.9-1+b1) ...
Processing triggers for man-db (2.8.5-2) ...
sauron [mc]$ hash -r
sauron [mc]$ type yosys
yosys is /home/merlin/fpga/fomu-toolchain-linux/bin/yosys
sauron [mc]$ make
yosys -D PVT=1 \
	-p \
	"ghdl --std=08 ../sb_ice40_components.vhd blink.vhd -e; \
	synth_ice40 \
	-top Fomu_Blink \
	-json blink.json" 2>&1 | tee yosys-report.txt
ERROR: No such command: ghdl (type 'help' for a command overview)
sauron [mc]$ yosys --version
Yosys 7f5c73d58fd732a96e480083896cd73c722849ba (Fomu build) (git sha1 66613d5, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)
```

Ok, so there are 2 problems:
1) I can't put /home/merlin/fpga/fomu-toolchain-linux/bin/ first in my path, because it contains a bad python I don't want.  So, I put it last, and indeed I already has yosys in debian
2) the other yosys didn't work any better. I have fomu-toolchain-linux_x86_64-v1.5.6
- https://workshop.fomu.im/en/latest/requirements/software.html pointed to
- https://github.com/im-tomu/fomu-toolchain/releases/tag/v1.5.6

umarcor, 2021-01-03:
Oh, I'm sorry for the disturbances, but luckily the solution is easy: use the [nightly](https://github.com/im-tomu/fomu-toolchain/releases/tag/nightly) release of `fomu-toolchain`, instead of v1.5.6. The latest "stable" release was tagged in march, and the GHDL/VHDL features were not added yet. Those were contributed some months ago.

We should either update the docs for recommending installing nightly builds of the toolchain, or we should tag a new release. @xobs, @mithro, WDYT?

> I can't put /home/merlin/fpga/fomu-toolchain-linux/bin/ first in my path, because it contains a bad python I don't want. So, I put it last, and indeed I already has yosys in debian

You can add it to the path only for running these commands. For example: `PATH=path/to/fomu-toolchain/bin:$PATH make`. Then, you can use the "bad" Python in the context of the Fomu toolchain, but leave your global PATH untouched.

Alternatively, you should be able to remove the built-in Python and use the one from the system. In fact, nightly release for Linux should not have any Python included (see https://github.com/im-tomu/fomu-toolchain/issues/16#issuecomment-706814966). Therefore, this problem should be gone. Please, let us know.

marcmerlin, 2021-01-03:
ok, took a little while to build ( and https://github.com/im-tomu/fomu-toolchain/issues/23 ) but it's built and works perfectly.  
Indeed, it'd be good to either release one version or update the workshop to tell users about this, or many give a precompiled json until then?

umarcor, 2021-01-03:
Well, you took a slightly longer path, but you arrived in the end 😆.

> give a precompiled json until then?

Providing a precompiled version of the JSON file does not make much sense. That is the netlist output of Yosys and the input to nextpnr. There are already some pre-compiled bitstreams which users can program for understanding "*how to use a pre-built design*". At the same time, "*how to place and route a Yosys netlist with nextpnr*" is already tested on many places. The purpose of the HDL examples in this workshop is for users to see how they can modify Verilog/VHDL sources for describing a hardware designs where no CPU or SoC is involved. As a result, executing Yosys and reading the logs is the most important didactic outcome of these HDL examples. If we provided JSON netlists, users would not understand the difference between verilog, VHDL and mixed HDL examples. In practice, all of them do just blink the LED.

> it'd be good to either release one version or update the workshop to tell users about this

Let's keep this issue open as a remainder.

umarcor, 2021-01-03:
@marcmerlin, strictly, you don't need to use the fomu-toolchain. You can run the examples in this workshop with tools installed however you want, as long as they provide all the required features. That is, you could use the Yosys available on the system, and build ghdl-yosys-plugin as a module (shared library). Then, you would add `-m path/to/plugin.so` when calling Yosys. See [ghdl/ghdl-yosys-plugin](https://github.com/ghdl/ghdl-yosys-plugin).

I would not suggest doing so just for the purpose of reading/doing this workshop. But you might consider it if you expect to use these tools intensively in the future.

mithro, 2021-01-04:
@umarcor - One quick thing that should be fixed is that the first command in the above Makefile should have failed and it shouldn't have tried to run nextpnr.

PR #513 (2021-08-10): hdl: remove yosys-report.txt
- set default shell to bash
- add pipefail option

Fixes #412
commit eda16712 (2021-08-10): hdl/**/Makefile: enable early rules termination (#513)

- set default shell to bash
- add pipefail option

Fixes #412
closed 2021-08-10 by umarcor