im-tomu/fomu-workshop #505: More permissive license for .pcf files

vazhnov, 2021-07-24:
Hello!

Thank you for this project!

I want to copy .pcf files ([this directory|https://github.com/im-tomu/fomu-workshop/tree/master/pcf]) to another project, to support different types of FOMU boards: https://github.com/BrunoLevy/learn-fpga/pull/44
But target repository license is BSD-3-Clause License, and this repository license is Apache License 2.0.

I'm not sure about their compatibility.

Is it possible to allow to share these files under some very permissive license, like BSD, CC0 or public domain? I think a lot of projects may want these files.

umarcor, 2021-07-24:
IANAL, but I since none of Apache 2.0 or BSD-3-Clause are viral, you can mix them in a project and distribute them together, without any issue. See https://softwareengineering.stackexchange.com/questions/40561/is-bsd-license-compatible-with-apache.

xobs, 2021-07-24:
That was my understanding as well. I originally picked Apache 2.0 because it's like the BSD license, but with an explicit patent clause. As far as I'm aware, there is no restriction to mix the two. I believe it's even possible to use it in a GPL project without issue, though that is not legal advice.