im-tomu/fomu-workshop #519: investigate tinyusb failure

proppy, 2021-07-30:
As mentioned in https://github.com/hathach/tinyusb/issues/984 the mass storage tinyusb seems to be failing to enumerate.

It would be nice to have tested manually and also covered by github tests.

mithro, 2021-07-30:
It would be awesome to have this tested in Renode.

@mateusz-holenko and @PiotrZierhoffer - Any thoughts on how we should be doing this?

tcal-x, 2021-07-30:
@proppy, I did some testing using the TinyUSB `cdc_msc` example built at this commit:
```
commit 43cb5fabfd6d0fca749e9512cc2061deab0adafd (HEAD -> master, origin/master, origin/HEAD)
Merge: 9aba2434 a5581b10
Author: Ha Thach <thach@tinyusb.org>
Date:   Fri Oct 16 22:08:07 2020 +0700

    Merge pull request #538 from kasjer/kasjer/uac2
    
    audio: Add headset example
```

It works on my Fomus that have bootloader v2.0.3 and v2.0.3-55-ge928b86 (locally built).  I see both the mass storage as well as an ACM tty.

It does NOT work with a Fomu that has bootloader v1.9.1.

You can check which bootloader you have using `dfu-util -l`.

tcal-x, 2021-08-20:
I noted this in PR #511 -- I tested `cdc_msc` with the current master TinyUSB, and it does work with bootloader v2.0.3.

The path to the bitstream does need updating as per the PR.

proppy, 2021-08-24:
Closing this now that #511 is merged.

closed 2021-08-24 by umarcor

mithro, 2021-08-25:
We should get Renode running on TinyUSB master.

umarcor, 2021-08-25:
Shall we reopen this issue to keep track of it or create a new one?