im-tomu/fomu-workshop #524: consider exposing SB_LEDDA_IP to verilog and litex samples

proppy, 2021-08-03:
It would be nice to expose `SB_LEDDA_IP` (as in https://github.com/im-tomu/foboot/blob/master/hw/rtl/sbled.py#L93) to the verilog and litex samples, so that developers going thru the workshop can do more advanced led PWM manipulation.

umarcor, 2021-08-03:
While I think it would interesting to have examples about how to use `SB_LEDDA_IP`, I don't necessarily agree with that allowing "more advanced led PWM manipulation". My perception is that `SB_LEDDA_IP` is for less advanced manipulation, in the sense that it's a IP with some given functionality that the user can learn and use but not modify (it's a hard core). Conversely, driving the PWM signals directly allows any hardware/logic to be used.

proppy, 2021-08-03:
oh yes, you're right, maybe a better way to phrase it would have been to say that it would allow to introduce developers to register based manipulation (which, I guess, carry their own complexity when implemented in hardware/logic).

umarcor, 2021-08-03:
Absolutely agree. Since this workshop has a top-bottom approach, the register based manipulation feels very natural for all the readers coming from the "CPU architecture" level, down to custom hardware peripherals. 
I just wanted to stress that "complex" and "advanced" are very different depending on the point of view!