im-tomu/fomu-workshop #525: consider adding SB_RGBA_DRV PWM verilog examples

proppy, 2021-08-03:
It would be nice to show how to drive `RGB0PWM` `RGB1PWM` `RGB2PWM` using a clock divider, so achieve the same color cycling effect as the `risc-v` sample.

umarcor, 2021-08-03:
That is actually the example design in hdl/verilog/blink, hdl/vhdl/blink and hdl/mixed/blink. 

- VHDL: https://github.com/im-tomu/fomu-workshop/blob/master/hdl/vhdl/blink/blink.vhd#L44-L78
- Verilog: https://github.com/im-tomu/fomu-workshop/blob/master/hdl/verilog/blink/blink.v#L51-L81

Mixed examples are a combination of those: having the counter in one language and the instantiation of the SB_RGBA_DRV in the other.

The resulting color cycles are not exactly the same, though, because we did not necessarily care about using exactly the same counter size/period.

proppy, 2021-08-03:
Sorry for the lack of precision I was referring to driving the PWM in a pattern similar to https://github.com/im-tomu/fomu-workshop/blob/master/riscv-blink/src/main.c#L9 rather than blinking full blue, full red, full green at `clock >> 25`, `clock >> 24` and `clock >> 23` intervals like in https://github.com/im-tomu/fomu-workshop/blob/master/hdl/verilog/blink/blink.v#L51-L81

See below:
[image]

umarcor, 2021-08-03:
Note that bits 25, 24 and 23 compose a 3 bit word with `2**3=8` different values. Therefore, the current HDL examples do not blink full blue, full red and full green. Instead, they cycle through the 7 colors (the last one is "black").

I just run the hdl/vhdl/blink example on my Fomu and the sequence is: off (black), blue, red, magenta, green, cyan, yellow, all (white).

Given said that, I think it would be interesting to have this better explained in the docs for the less experienced HDL users. It would also be nice to have some more creative examples that show how to get other colour effects.

proppy, 2021-08-03:
Oh yes, sorry for the miss-understanding, what I meant was that it blinks red, blue and green with between 100% duty cycle and 0% duty cycle at consecutive 2^ interval (causing the 3 components to get mixed as you described), I'd be interesting to provide examples that modulate the PWM duty cycle to get access to more colors.