im-tomu/fomu-workshop #583: VHDL examples don't build with fomu-toolchain

patrislav1, 2021-11-29:
When using [fomu-toolchain](https://github.com/im-tomu/fomu-toolchain), the [hdl/vhdl](https://github.com/im-tomu/fomu-workshop/tree/master/hdl/vhdl/blink) and [hdl/mixed](https://github.com/im-tomu/fomu-workshop/tree/master/hdl/mixed/blink) examples fail to build (see below).

However, the build succeeds when using the toolchain from the [OSS CAD Suite](https://github.com/YosysHQ/oss-cad-suite-build) instead.

```
$ PATH=../../../fomu-toolchain-Linux/bin:$PATH FOMU_REV="pvt" make 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+3619 (open-tool-forge build) (git sha1 c403c984, gcc 9.3.0-17ubuntu1~20.04 -Os)

-- Running command `read -define PVT=1' --

-- Running command `ghdl --std=08 ../../sb_ice40_components.vhd blink.vhd -e; 	synth_ice40 	-top Fomu_Blink 	-json blink.json' --

1. Executing GHDL.
warning: ieee library directory '/__w/1/s/_packages/build_linux_x86_64/fpga-toolchain/lib/ghdl/ieee/v08/' not found
error: cannot find "std" library
ERROR: vhdl import failed.
make: *** [Makefile:39: blink.json] Error 1
```

umarcor, 2021-11-29:
@patrislav1, when using the fomu-toolchain, did you set GHDL_PREFIX as explained in https://github.com/YosysHQ/fpga-toolchain#using-ghdl ?

patrislav1, 2021-11-29:
When I set `GHDL_PREFIX`, I get the same error.

umarcor, 2021-11-29:
I suggest you try executing `./.github/tests.sh` (https://github.com/im-tomu/fomu-workshop/blob/master/.github/tests.sh), which is the script used in CI: https://github.com/im-tomu/fomu-workshop/blob/a4c866f35c9e5229642f441d23529e93f633fdbc/.github/workflows/test.yml#L61-L65

patrislav1, 2021-11-29:
tests.sh runs, but then complains about `zig` missing ;-)

`hdl-tests.sh` (which is the one that actually tests the VHDL examples) raises the same "ieee library directory not found" error as above, as soon as it reaches the VHDL example

```
$ .github/hdl-tests.sh
::group::Verilog Blink example
+ cd verilog/blink
+ make FOMU_REV=pvt blink.bit
make: 'blink.bit' is up to date.
+ '[' -f blink.bit ']'
::endgroup::
::group::Verilog Blink (expanded) example for Hacker board
+ cd verilog/blink-expanded
+ make FOMU_REV=hacker blink.bit
make: 'blink.bit' is up to date.
+ '[' -f blink.bit ']'
::endgroup::
::group::Verilog Blink (expanded) example for PVT board
+ cd verilog/blink-expanded
+ make FOMU_REV=pvt blink.bit
make: 'blink.bit' is up to date.
+ '[' -f blink.bit ']'
::endgroup::
::group::VHDL Blink example
+ cd vhdl/blink
+ make FOMU_REV=pvt blink.bit

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+3619 (open-tool-forge build) (git sha1 c403c984, gcc 9.3.0-17ubuntu1~20.04 -Os)

-- Running command `read -define PVT=1' --

-- Running command `ghdl --std=08 ../../sb_ice40_components.vhd blink.vhd -e; 	synth_ice40 	-top Fomu_Blink 	-json blink.json' --

1. Executing GHDL.
warning: ieee library directory '/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ieee/v08/' not found
error: cannot find "std" library
ERROR: vhdl import failed.
make: *** [Makefile:45: blink.json] Error 1
```

umarcor, 2021-11-29:
Can you please check `/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ieee/v08/` or whichever location where the fomu toolchain was extracted? Are there any VHDL sources at all? You should find, at least, https://github.com/ghdl/ghdl/tree/master/libraries/std and https://github.com/ghdl/ghdl/tree/master/libraries/ieee.
However, I think you are missing `ghdl` between `lib` and `ieee`.

patrislav1, 2021-11-29:
The `GHDL_PREFIX` is correct, but somewhere down the rabbit hole it gets corrupted. I'm not sure where the other path, that `ghdl` ends up using, is coming from.

```
$ export GHDL_PREFIX=$(dirname $(ghdl --libghdl-library-path))
$ echo $GHDL_PREFIX
/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib
$ find $GHDL_PREFIX -name std
/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ghdl/std
/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ghdl/src/std
$ find $GHDL_PREFIX -name ieee
/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ghdl/ieee
/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ghdl/src/ieee
$ find $GHDL_PREFIX -name "*.vhd*" | wc -l
71
$ FOMU_REV="pvt" make                          

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+3619 (open-tool-forge build) (git sha1 c403c984, gcc 9.3.0-17ubuntu1~20.04 -Os)

-- Running command `read -define PVT=1' --

-- Running command `ghdl --std=08 ../../sb_ice40_components.vhd blink.vhd -e; 	synth_ice40 	-top Fomu_Blink 	-json blink.json' --

1. Executing GHDL.
warning: ieee library directory '/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ieee/v08/' not found
error: cannot find "std" library
ERROR: vhdl import failed.
make: *** [Makefile:39: blink.json] Error 1
$ ls /home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ieee/v08/
ls: cannot access '/home/patrick/src/fomu-workshop/fomu-toolchain-Linux/lib/ieee/v08/': No such file or directory
```

patrislav1, 2021-11-29:
> However, I think you are missing `ghdl` between `lib` and `ieee`.

Yes, that was the reason. Thanks.

Would it beneficial to add something in the Makefile that sets `GHDL_PREFIX` to `$(dirname $(ghdl --libghdl-library-path))/ghdl`?

umarcor, 2021-11-29:
> Would it be beneficial to add something in the Makefile that sets `GHDL_PREFIX` to `$(dirname $(ghdl --libghdl-library-path))/ghdl`?

That might be worth a try. However, note that the CI includes three different test environments, aparto from fomu-toolchain:

- An all-in-one container.
- Fine-grained containers.
- MSYS2 packages.

On MSYS2 (Windows), GHDL_PREFIX is not necessary, but I don't know whether setting it incorrectly can have undesired effects.

On the containers, the location should be the same as the PREFIX used when building GHDL. Hence, GHDL_PREFIX should not be necessary. Again, I don't know whether setting it incorrectly can have undesired effects.

closed 2021-11-29 by umarcor